j7126/OctoPrint-Dashboard #412: [Bug]: Layer count start a zero not 1

nicval, 2024-09-13:
### The problem

layer displayed is always 1 layer off
prusa start at layer 1 
display layer progress start at layer 1
dashboard start at layer 0 so all the layer count is offset by 1 layer
so if i have a gcode that is 52 layers dashboard count to 51 layers

### Version of OctoPrint

1.10.2

### Platform

Raspberry Pi 4

### Operating system running OctoPrint

OctoPi

### Dashboard version

1.19.11

### Python version

3.92

### Printer model & used firmware incl. version

Prusa Mini+ 6.2.0-alpha1

### Browser and version of browser, operating system running browser

Brave Version 1.70.107

### Logs and other files needed for analysis

- [X] Systeminfo Bundle (always include!) See [here](https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887) if you don't know where to find that.
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### File uploads & additional information

[octoprint-systeminfo-20240913014846.zip](https://github.com/user-attachments/files/16989645/octoprint-systeminfo-20240913014846.zip)

gmarch, 2024-10-05:
I was just going to submit the same issue.

[image]

j7126, 2024-10-06:
What slicer are you using?

Can you please upload a GCODE file that triggers the issue?

nicval, 2024-10-06:
it is prusa slicer and does it with all gcode from prusaslicer

nicval, 2024-10-06:
[Gift Box Prank.zip](https://github.com/user-attachments/files/17268181/Gift.Box.Prank.zip)

if you try tis one it always one layer shifted it is like it start at layer 0 and not a layer 1
a few updates back and it was working perfectly

the problem for me started at version 1.19.1 in my testing when you fixed bug #303 
Fix total layer number wrong https://github.com/j7126/OctoPrint-Dashboard/issues/303

j7126, 2024-10-06:
I have just released a new version 1.19.12 which should fix the issue.

nicval, 2024-10-06:
it solves the problem thanks

closed 2024-10-06 by nicval